Renaud11232/OctoPrint-Marlin-Flasher #2: user: Current not implemented on linux/arm

Renaud11232, 2019-04-02:
There might be an issue with `arduino-cli` I've opened an issue.

While waiting for official help on their part, I've found a workaround : 

On Linux: open the `/etc/init.d/octoprint` with your favorite editor. Find the following lines : 
```
       start-stop-daemon --start --background --quiet --pidfile $PIDFILE --make-pidfile \
       --exec $DAEMON --chuid $OCTOPRINT_USER --user $OCTOPRINT_USER --umask $UMASK --nicelevel=$NICELEVEL \
       -- serve $COMMAND_ARGS $DAEMON_ARGS
```
and change it for
```
       start-stop-daemon --start --background --quiet --pidfile $PIDFILE --make-pidfile \
       --exec /usr/bin/env USER=$OCTOPRINT_USER $DAEMON --chuid $OCTOPRINT_USER --user $OCTOPRINT_USER --umask $UMASK --nicelevel=$NICELEVEL \
       -- serve $COMMAND_ARGS $DAEMON_ARGS
```

The issue does not appear on windows afaik

AndKe, 2019-04-03:
thank you, I just got here to report this..  :)  (OctoPi)

FrancYescO, 2019-04-21:
looks like your "search for" lines are wrong:
this is what i got
```start-stop-daemon --start --background --quiet --pidfile $PIDFILE --make-pidfile \
       --exec $DAEMON --chuid $OCTOPRINT_USER --user $OCTOPRINT_USER --umask $UMASK -- serve $DAEMON_ARGS```

Renaud11232, 2019-04-21:
The lines I described in the original post are direcly copied from the file I have.
I manually installed OctoPrint on an OrangePi on armbian.
Anyway, in your case that should become
```
start-stop-daemon --start --background --quiet --pidfile $PIDFILE --make-pidfile \
       --exec /usr/bin/env USER=$OCTOPRINT_USER $DAEMON --chuid $OCTOPRINT_USER --user $OCTOPRINT_USER --umask $UMASK -- serve $DAEMON_ARGS
```

FrancYescO, 2019-04-21:
i copied what was in your first comment, and after installing liquidcrystl lib i got my first successfull flash :)

M-Gregoire, 2019-05-29:
Thanks a lot, this works perfectly!

Renaud11232, 2019-08-27:
This [issue](https://github.com/arduino/arduino-cli/issues/182) is resolved in newer versions of arduino-cli.

BUT, don't update yet, newer versions won't be compatible until I publish a fix for #12

closed 2019-08-27 by Renaud11232