Renaud11232/OctoPrint-Marlin-Flasher #5: Really Cool

Eddiiie, 2019-04-05:
So I have used this and it is really cool.  The ONLY thing I would love is the ability to edit the sketch (configuration.h in my case) ON the Pi, somehow.  Still a PITA to load the sketch on my Windows box make an edit, save it, zip it, upload it, flash it... 

Nonetheless, this is a cool plugin for a cool package.  Thank you!

Renaud11232, 2019-04-05:
Hello,

First of all I'm glad to hear (or read) you like it :)
I thought about adding the ability to edit config files directly within Octoprint but it will break some things.
It can/will break compatibility with : 
* Any non-Marlin code as the files `Configuration.h` and `Configuration_adv.h` won't be there.
* The `2.x.x` branch of Marlin as the file structure totally changed so the configuration files are not located at the same place.

I may give it a bit more time to think about a clean way to implement it.

mikekscholz, 2019-04-05:
Does the entire library NEED to be zipped in order to function? Is it possible to just have it uncompressed in a directory so people can just text edit whatever files they may need to?

Eddiiie, 2019-04-06:
Well, good point about newer generations of firmware/controllers.  I am moving to 32 bit processor (Arduino Due) aka firmware version 2.0 (Marlin and Repetier) as soon as my Smart RAMPs comes in from China...  I wont be on 1.x much longer.   

At the same time, it would be great timing to have access to the 'files' to make changes to dial the printer in with the new controller hardware.

Renaud11232, 2019-04-06:
@mikekscholz At the moment it only accepts `.zip`files or precompiled `.hex`files.

@Eddiiie If I understad  well, you'd like to have a way to directly edit firmware files into the web UI ?

Anyway I will close this issue and reopen two one for each of your suggestions.

closed 2019-04-06 by Renaud11232