Renaud11232/OctoPrint-Marlin-Flasher #8: I'm lost

mark-hahn, 2019-04-13:
The plugin UI on my octoprint doesn't seem to match anything shown in the readme for this repo.  This is what I see ...

```
Marlin Flasher Settings
Path to arduino-cli
Sketch .ino
Maximum sketch size 20 MB  *requires restart
```

Renaud11232, 2019-04-13:
Hello,

The page you see is the settings page. Where you can set up the path where you installed `arduino-cli`. The name of the sketch `.ino`file. and the maximum upload file size. This page is located under `Settings > Marlin Flasher`.

The main page of the plugin is accessible via the main screen of octoprint's UI. It should be a tab located after the `Temperature`, `Control`, `GCode Viewer` and `Terminal` tabs.

Let me know if it helped so I can mark this as solved.

maxlinux2000, 2019-04-19:
Hello,
I'm trying to make it work, and I've installed arduino-cli in / usr / local / bin (the version for arm 32bit)

``root @ orangepipc: ~ # ls / usr / local / bin / arduino-cli
/usr/local/bin/arduino-cli``
And set up this path in the configuration.

My demo board pc is OrangePi PC with armbian5.75 (ubuntu 18.04) and work just fine with the last octoprint stable version (Version 1.3.10)

But now I do not know what I have to do ...

I have assumed that I have to compile on my laptop with arduino-ide and save the .hex file (Marlin.ino.mega.hex in my case)
I have uploaded this file with Marlin Flasher, a message tells me that it has been uploaded successfully, but then it does not appear anywhere.
That I have to do?

What we really need is a mini-tutorial that explores how to install and configure the plugin

maxlinux2000, 2019-04-19:
I also tried to upload the sketch compressed in Marlin's .zip (1.1.9 bufix for gt2560) which is the same one I'm using in my 3d printer.
I compressed the Marlin folder in Marlin.zip, I upload the file, a green message comes out that tells me that the Marlin.ino has been successfully uploaded, but then it does not appear in the list below.

Where am I wrong?

Renaud11232, 2019-04-20:
Hello,

Sorry if some things are unclear I wrote the plugin so for me everything was obvious when writing the README. So I'll make a first answer here and then I'll try to update the README file and maybe create a wiki (never tried before on github) to make things more clear. With a step by step guide/tutorial as you suggested or something similar.

Anyway I'll try to make a summary on how I designed the plugin and how it's meant to be used.

The plugin is split in two pages:
*  The configuration page where you *most likely*¨will only have to change the path to `arduino-cli`. Here you can also set the name of the main firmware `.ino` file. As you are using Marlin, `Marlin.ino` is the default value and should work. And the maximum file upload size. The default value should also work fine for Marlin.

* The main UI page where you can flash precompiled firmware with `.hex` files or compile then flash zipped firmware source code. If using a `zip` file,

The flashing process goes as follows:

1. Upload your `.hex`/`.zip` file in the **Sketch** section of the main page
2. In order to be able to sucessfully flash your board you will first need to install the required tools and board definitions.These are named cores in `arduino-cli`. This is done through the **Cores** section. There you can search for cores to install (or remove). `arduino-cli` does **NOT** come with the basic boards preinstalled. So even the Arduino Mega 2560 won't be installed until you manually install it. To install them search for `Arduino` and install the `arduino:avr` core.
3. You might also need to install some libraries in order to sucessfully compile the firmware. This is done the same way you install the cores but through the **Libraries** section.
4. Select the appropriate board in the **Flash** section.
5. If the selected board requires it, select other board options. They should appear right under the board selection
6. Hit the **Flash** button and wait.

If your board/libraries are not available, you can also install them in `~/Arduino/libraries` for the libraries and `~/Arduino/hardware` for the boards, the same way you would install them on Arduino IDE.

If I understood you problem corectly it seems there are no cores installed as you cannot select the board. If you usually compile for an `Arduino Mega 2560`, installing the `arduino:avr` core as described above should be enough.

I hope it made things a bit more clear, if not feel free to come back for help.

Renaud11232, 2019-04-20:
As I now have pushed an updated readme with a link to the wiki i will close this issue. If you see any problem with the wiki, feel free to open a new issue. I hope it will help you and other people that might have problems figuring out how the plugin works :)

maxlinux2000, 2019-04-20:
/home/max/Vídeos/MarlinFlasher-2019-04-20_21.27.54.mkv

maxlinux2000, 2019-04-20:
Hello,
I am trying to follow the document, but there is something missing according to my opinion. Look at the video that I recorded, please

https://youtu.be/jqR1gmkNxuQ

tnx

Renaud11232, 2019-04-20:
Looking at the video you sent, it seems everything is configured properly however it looks like you didn't install any core.
First of all can you try to install the `arduino:avr` core. To do so, go in the `Cores` section. in the search field type in `arduino` and hit the `Search` button.
After a few seconds it should come with a list similar to this one
[image]

Once you have that, click on the install button (the one with the little download icon) on the line where it says `arduino:avr` in the `ID` field (The order of the search results can change). This should take a bit of time depending on your connection speed and SD card speed.

Once that is installed you should now have all the Arduino/Genuino boards available. This process is documented now on the [wiki](https://github.com/Renaud11232/OctoPrint-Marlin-Flasher/wiki/Usage#cores)

Did that solve the problem ?

FrancYescO, 2019-04-21:
[image]

Renaud11232, 2019-04-21:
@FrancYescO check #2

closed 2019-05-12 by Renaud11232