Renaud11232/OctoPrint-Marlin-Flasher #11: Improve error messages

Renaud11232, 2019-05-06:
Some error messages needs to be more clear
ie : The printer is not ready

pattonwebz, 2019-06-14:
I also had issues figuring out what some errors meant but I expect that in a few cases that is because `arduino-cli` doesn't pipe back any info - it just silently fails.

pattonwebz, 2019-08-28:
@Renaud11232 I see this one was closed by the other commit but isn't that just an update on a single message? Would there not be other messages that could be improved as part of this one single larger issue thread?

pattonwebz, 2019-08-28:
I do not have other messages that I know of. My errors were silent lol

I just figured maybe there were other messages that could get some TLC while you were looking at them, if there's not any more that are really problematic then awesome :)

pattonwebz, 2019-08-28:
I do not remember the exact time I had problems or how to trigger it. 

I was able to work around it by just going to command line and compiling and flashing manually. When I done that I discovered that at certain points `arduino-cli` did not give any errors - just failed silently. I first noticed this during compile when I was missing the `U8glib.h` library for my display. I did wonder if this was the same thing that made it not work for me though the plugin GUI but I never confirmed that.

I am unsure how you could account for errors that are silent fails on cli tools though, perhaps maybe just a timeout and if no success is shown in a certain time then say that it probably failed? It seems like with that idea I would just be introducing more confusing error messages rather than actually making any clearer :confused: 

Sorry to have got you to reopen this one for me when really I had no actual value to provide here for a solution (or to even guide to a problem).

Renaud11232, 2019-08-28:
indeed it is on a single message but if you have other messages that seems unclear i'll reopen the issue and try to fix them

Renaud11232, 2019-08-28:
Do you remember when/how you got that ?

Renaud11232, 2019-08-28:
No need to be sorry.
But if arduino-cli doen't give information itself I don't thing I can improve that by myself.
I'll reclose the issur then.

commit 3ff3e134 (2019-08-27): improved error message
fixes #11
closed 2019-08-28 by Renaud11232