Renaud11232/OctoPrint-Marlin-Flasher #30: Exception when searching for Cores

platinum95, 2019-11-02:
After installing the plugin and setting up arduino-cli, I'm unable to search for cores.
The search button remains greyed out showing "Searching" indefinitely. The following exception shows up in the OctoPrint log:
```
octoprint - ERROR - Exception on /plugin/marlin_flasher/cores/search [GET]
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 1149, in decorated_view
    return no_firstrun_access(flask_login.login_required(func))(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 1168, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask_login.py", line 758, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask_principal/__init__.py", line 198, in _decorated
    rv = f(*args, **kw)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_marlin_flasher/__init__.py", line 112, in search_cores
    return flask.make_response(flask.jsonify(result), 200)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/json.py", line 237, in jsonify
    return current_app.response_class(dumps(dict(*args, **kwargs),
ValueError: dictionary update sequence element #0 has length 7; 2 is required
2019-11-02 21:13:25,382 - tornado.access - ERROR - 500 GET /plugin/marlin_flasher/cores/search?query=arduino (::ffff:192.168.1.17) 18776.28ms
```

platinum95, 2019-11-02:
After a bit of digging around, seems like Knockout on the JS side is expecting a list, which is passed to `flask.jsonify`. Serialising lists using flask's jsonify is only supported in Flask v0.11 onwards, but for whatever reason version 0.10 was installed on my device when OctoPrint was set up. I've managed to temporarily get around this issue by using `json.jsonify` to serialise the list, and pass the resultant string into `flask.Response`.

Renaud11232, 2019-11-02:
Hello,
The client side indeed expects a json array.  and that's what's (normally) returned by arduino-cli.
And indeed I now see that version <=0.10 do not suport array/list serialization.

Octoprint requires flask<0.11, now I wonder why it worked on my dev environment as well as on my OrangePi.

I'll investigate that tomorrow, and try to publish a fix.

platinum95, 2019-11-02:
After sourcing into the OctoPrint venv, I was able to force-install flask v0.10.1 with pip which seems to have list-serialisation support, so that's also a potential fix. Not sure why 0.10 was installed when 0.10.1 was also valid for the OctoPrint libs.

Renaud11232, 2019-11-03:
Oh I assumed you were on 0.10.1 and not 0.10. Because Octoprint itself requires >=0.10.1. So i'm usure how that happened either.

FrancYescO, 2019-12-01:
having same issue on 0.3.0, any easy step to install the rc where i suppose it got fixed?

Renaud11232, 2019-12-01:
The fix should be included in release 0.3.0, the update should be available directly withint Octoprint's software update system.

Just to be sure it is the same issue, can you give me the output of these commands ? (Guessing you are running OctoPi)

```bash
~/oprint/bin/pip show flask
~/oprint/bin/pip show OctoPrint-Marlin-Flasher
```
And also octoprint's logs

Thanks

FrancYescO, 2019-12-01:
[octoprint.log](https://github.com/Renaud11232/OctoPrint-Marlin-Flasher/files/3908553/octoprint.log)
```
pi@octopi:~ $ ~/oprint/bin/pip show flask
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
Name: Flask
Version: 0.10.1
Summary: A microframework based on Werkzeug, Jinja2 and good intentions
Home-page: http://github.com/mitsuhiko/flask/
Author: Armin Ronacher
Author-email: armin.ronacher@active-4.com
License: BSD
Location: /home/pi/oprint/lib/python2.7/site-packages
Requires: itsdangerous, Werkzeug, Jinja2
Required-by: Flask-Principal, Flask-Login, Flask-Babel, Flask-Assets
pi@octopi:~ $ ~/oprint/bin/pip show OctoPrint-Marlin-Flasher
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
Name: OctoPrint-Marlin-Flasher
Version: 0.3.0
Summary: Plugin that allows you to flash your printer with the latest Marlin version, or Arduino code
Home-page: https://github.com/Renaud11232/Octoprint-Marlin-Flasher
Author: Renaud Gaspard
Author-email: gaspardrenaud@hotmail.com
License: MIT
Location: /home/pi/oprint/lib/python2.7/site-packages
Requires: OctoPrint, pyduinocli, intelhex, marshmallow, flask
Required-by:
```

Renaud11232, 2019-12-01:
I think this is not the same issue, I opened a new one (#34)  can you provide me the output of
```bash
~/oprint/bin/pip show pyduinocli
```
On that new issue ?

commit 2c031e15 (2019-11-05): fixes #30
closed 2019-12-01 by Renaud11232