Renaud11232/OctoPrint-Marlin-Flasher #32: octoprint stuck on Marlin Flash

idioffo89, 2019-11-30:
Hi, I have my octoprint main page stuck on Marlin Flasher wizard page, the note on it says that I should be able to continue by clicking finish but I have tried nothing I still on the wizard page thus can't use octoprint at the moment.

How I can disable or uninstall Marlin Flash maybe through command line so I can at least access my octoprint? I have attached a picture so you can see where I'm stuck. Any help will be appreciated.

[image]

Renaud11232, 2019-11-30:
Hello,
I have a few questions : 

Which version did you install ?
If you don't know, do you know when/how you installed it ?

Which browser are you using ?

And what system are you running it on ? (Windows, Linux, Raspberry...) ?

Thanks

idioffo89, 2019-11-30:
So I have installed it a while ago, and I do know I'm not running the latest version of Marlin Flasher because I have OctoPod app on my phone as that is the only way I can use octoprint right now and it does tell me that there is a new version of Marlin Flasher.

I'm using Safari as my main browser but I did try on chrome and firefox, I'm still unable to go pass the wizard page

And I'm running octoprint on a Raspberry Pi

Renaud11232, 2019-11-30:
Thanks,

You can uninstall it using commandline :
```bash
~/oprint/bin/pip uninstall OctoPrint-Marlin-Flasher
```
This might will probably need Octoprint to be restarted.

But just before you do that, can check a few things, so I can know why it does that.
Can you check in your browser console if you have any error when you press the finish/next button
On Chrome, press F12 and go to the console tab.

It might also be useful if you could send your octoprint logs.

Thank you and I hope it helps

idioffo89, 2019-11-30:
yes of course I can do that. Can you tell me how to get the octoprint logs? so I can get that and send it over

So here is what I'm seeing on the console side
[image]

Renaud11232, 2019-11-30:
Thank you,

The logs can be found in `~/.octoprint/logs`
There should be one named `octoprint.log`.

Also, it seems there's one error occuring before the screenshot you gave. Is it the same one repeating or is it a different one ?

idioffo89, 2019-11-30:
Yes it was the same error I just try it multiple times to make sure I wasn't seeing any other error.

Here is the log

[image]

idioffo89, 2019-11-30:
I'm unable to use the command line you sent me to uninstall Marlin Flasher. I was able to get to the bin folder, I do see PIP in the list but trying to get to it I get an error saying pip: not a directory. Am I missing something?

Renaud11232, 2019-11-30:
Did you type 
```
~/oprint/bin/pip uninstall OctoPrint-Marlin-Flasher
```
or
```
cd ~/oprint/bin/pip uninstall OctoPrint-Marlin-Flasher
```
Indeed, pip is not a directory it's an executable.

If you are in the `~/oprint/bin/` folder you can type
```
./pip uninstall OctoPrint-Marlin-Flasher
```
But the first command should work as is, no matter the folder you currently are in.

I hope I'm clear :/

idioffo89, 2019-11-30:
Ah got it thanks, I was trying to cd into the folder first. I was able to get it.

Thank you again for your help with this

Renaud11232, 2019-11-30:
No problem,

I'll close this.

closed 2019-11-30 by Renaud11232