Renaud11232/OctoPrint-Marlin-Flasher #34: pyduinocli version mismatch

Renaud11232, 2019-12-01:
This is recreated from #30 
I suspect 0.3.0 does not support the newest pyduinocli.

FrancYescO, 2019-12-01:
```
Name: pyduinocli
Version: 0.4.1
Summary: Wrapper library around arduino-cli
Home-page: https://github.com/Renaud11232/pyduinocli
Author: Renaud Gaspard
Author-email: gaspardrenaud@hotmail.com
License: UNKNOWN
Location: /home/pi/oprint/lib/python2.7/site-packages
Requires:
Required-by: OctoPrint-Marlin-Flasher
```

Renaud11232, 2019-12-01:
If it's what i think it is, this should now be fixed.
Can you check if 0.3.1 fixed it ?

FrancYescO, 2019-12-01:
should be in prod? the check updates says all up to date

FrancYescO, 2019-12-01:
ehm, MarlinFlasher totally disappeared from my octoprint

Renaud11232, 2019-12-01:
Well, well, well.
Can you restart OctoPrint/reload the page, and resend the logs ?

FrancYescO, 2019-12-01:
after another reboot after the one after MarlinFlasher upgrade, it come back and board selection and search is working :)

Renaud11232, 2019-12-01:
Oh so it does work. Nice
I'll close this then.

Thanks for your patience

closed 2019-12-01 by Renaud11232