Renaud11232/OctoPrint-Marlin-Flasher #36: Donations

ForsakenNGS, 2020-01-19:
TLDR: Do you accept donations / where can I tip you?

Long story: I'm rebuilding my first self-built 3d printer into a ultra-low-duty CNC, which requires a lot of firmware tweaking right now. Not having to replug the board all the time is really a godsend. For the amount of time you save me I'd love to give you a small tip and probably others feel the same way, too.

Renaud11232, 2020-01-20:
Hello,
I'm really glad you like the plugin :)
This is why I created it, I wanted a way to upgrade the firmware without the need of plugging it to another computer.

That being said, I currently don't have anything setup for donations/tips simply because I didn't even thing about it.

I might give it a try. Without any obligation of course.

Renaud11232, 2020-01-22:
Hello,
I've added a [image] button in the readme.
I just have to thank you then, if you still want to donate, of course, this is not an obligation

closed 2020-01-22 by Renaud11232

ForsakenNGS, 2020-02-12:
It took me a while, but better late then never. Have a drink on me :)