Renaud11232/OctoPrint-Marlin-Flasher #37: i'm getting "board option fetch failed"

mark-hahn, 2020-01-29:
I have the sanguino board (core) loaded.  I got the installed message.  I put up a hex file.  Now I'm getting "board option fetch failed".  Any idea what could cause this?

Renaud11232, 2020-01-29:
Hello,
Did you install the `arduino:avr` core package ?
If not this might be the cause.
Tell me if that did the trick.
I'll look into improving that message.

mark-hahn, 2020-01-29:
Thanks, that fixed it.  One problem I had is that there was no feedback when a core was being loaded.  The download button didn't indicate it was doing anything.  I clicked around a few times and nothing seemed to happen.  Later the installed dialog seemed to come from nowhere.

Thanks again.  I uploaded firmware and I'll go test it now.

closed 2020-01-29 by mark-hahn

Renaud11232, 2020-01-30:
I just added loading animations when the install/uninstall buttons are clicked for the cores and libraries.
I also added one while the board options are loading.
(Not released yet)
I opened #38 regarding the error messages being unclear.

Both improvements should be available in the next release.

mark-hahn, 2020-01-30:
Thanks mucho.