Renaud11232/OctoPrint-Marlin-Flasher #40: Using Platform.io configuration on SKR 1.3 board

GithStreamer, 2020-02-04:
I have followed the instructions and installed the Marlin Flasher as well as the platformio components.  When I drop the zip file into the Marlin Flasher in Octoprint, it takes a moment to compile.  It says Uploading and gets about half way through the progress bar and then says - /dev/ttyACM0/firmware.bin: Not a directory - Failed.  I have tried looking around but am drawing a blank.

Renaud11232, 2020-02-04:
This might be a bug.
I recently got an SKR 1.3 too but didn't had the time to replace my current board.
Once I mount it on my printer, I'll be able to try to look into this.

closed 2020-02-18 by Renaud11232

Renaud11232, 2020-02-18:
Hello,
In order to get it working you have to install an additional piece of software.
The [wiki](https://github.com/Renaud11232/OctoPrint-Marlin-Flasher/wiki/PlatformIO-Installation#armmbed-boards-linux-only) should help to know how to get it working.

commit b2082ce0 (2020-02-06): now letting platformio detect the port to upload firmware
fixes #40