Renaud11232/OctoPrint-Marlin-Flasher #55: SKR question

TwoToneEddy, 2020-08-19:
Hi,
I'm trying to get your plugin working with my SKR V2.0 and I have a few questions as I found some of the instructions abit unclear.
For context i'm using a linux PC for the coding and a raspberry pi 3 running octoprint that I connect to via ssh.

You mention installing platfomio, should this be done on both my linux machine and the octoprint machine? I'm aware you need platformio to compile the binary / hex files so I have this installed on my main linux machine. This works fine and I can compile a binary file which appears to work when i put it on the SD card of the PI manually.

Does this plugin work by flashing a single file from the client machine (my main linux machine) or does it compile it on the pi?

Many thanks
Lee

Renaud11232, 2020-08-19:
Hello,

Platformio-Core to be installed on the machine running OctoPrint.

Then you need to upload a zip file containing the source code of Marlin. The plugin will then compile it and upload it to the board.

I hope it helped

TwoToneEddy, 2020-08-19:
I understand, thanks for the explanation

closed 2020-08-19 by TwoToneEddy