Renaud11232/OctoPrint-Marlin-Flasher #56: Allow flashing from within itself

girrrrrrr2, 2020-08-29:
Basically for those of that use git to help organize our marlin firmwares it would be nice to be able to flash marlin from within octoprint instead of having to do it from a computer. I know its not a major priority but it would be nice to be able to do.

Renaud11232, 2020-08-29:
Hello,

If I understand correctly (correct me if i'm wrong) you'd want to be able to use a git repository URL instead of having to upload a zip file each time you want to flash ?

If that is the case, don't expect this too soon. But I might give it a go.

girrrrrrr2, 2020-08-29:
Yeah that would be perfect!
Thanks for the quick response! 
On Saturday, August 29, 2020 Renaud11232 <reply@reply.github.com> wrote:

Hello,

If I understand correctly (correct me if i'm wrong) you'd want to be able to use a git repository URL instead of having to upload a zip file each time you want to flash ?

If that is the case, don't expect this too soon. But I might give it a go.

AvanOsch, 2020-11-02:
+1
This would be pretty awesome!
Just an "install from URL" field (like in the Octoprint Plugin Manager) would be great.

This would also not limit the user to using Git.
I can also imagine it's easier to just point to a zip file.

commit 3f54f812 (2021-02-07): adds a way to download firmware via a URL
fixes #56
closed 2021-04-01 by Renaud11232