Renaud11232/OctoPrint-Marlin-Flasher #57: Not able to install

lpvezina, 2020-10-10:
I'm struggling configuring the plug-in.  I'm running Octoprint on a Raspberry PI.  What I've done so far from Windows:
- Downloaded arduino-cli_0.13.0_Linux_ARMv6.tar.gz
- Extract the gz and then the tar => ended up witht he file arduino-cli
- Uploaded this file to my Raspberry PI and placed it in a new folder I've created: /home/pi/arduino
- Went to the MArlin Flasher Plug-in configuration and set the path to /home/pi/arduino/arduino-cli

After saving, I'm having the following error message: The configured path does not point to an arduino-cli executable

What am I doing wrong?

Renaud11232, 2020-10-14:
Hello,

`arduino-cli@0.13.0` is currently not supported by the plugin, Since 0.11, they have introduced some breaking changes, and I need to adapt my plugin. In the meantime, you can use `0.10.x`.

Another thing to check are the permissions on the `arduino-cli` executable. Is it actually executable. To be sure, you can type `chmod a+x arduino-cli` in a console.

Let me know if it helped.

lpvezina, 2020-10-14:
Shame on me, I didn't saw your warning about the Arduino-cli version on the wiki page :(  I'll give it a try tonight.

Renaud11232, 2020-11-02:
Hello,

Did it work ?

lpvezina, 2020-11-10:
I finally took the time to try it back and once I copied it, and after I set the permission for execution, it worked.  I'll try to flash it hopefully this weekend. I'm closing this support request. Thanks!

closed 2020-11-10 by lpvezina