Renaud11232/OctoPrint-Marlin-Flasher #60: Clarification of the firmware compilation process

narfel, 2020-11-18:
Very new to this, so sorry if this is obvious. The scope of this plugin is to compile and upload on the octoprint host via cli? Is there a way to upload and flash just a bin that was compiled on a different machine?

narfel, 2020-11-18:
I think this answer by @AvanOsch is quite the same than what i ask for. 

 @AvanOsch in https://github.com/Renaud11232/OctoPrint-Marlin-Flasher/issues/56#issuecomment-720703295_

Renaud11232, 2020-11-28:
Hello,

Sorry for the late answer.

Currently, using Arduino-cli you can upload and flash either a zip file containing the source code or a compiled `.hex` file.

Using platform.io, you can only use the source code.

I hope this helps

closed 2021-02-16 by stale[bot]