Renaud11232/OctoPrint-Marlin-Flasher #62: The configured path does not point to an arduino-cli executable.

kd7eir, 2020-11-30:
The configured path DOES point to the arduino-cli executable, it's version 0.10.0, I can call it from a command prompt and use it. Is there some magic incantation needed to make this plugin actually see the file?

Renaud11232, 2020-12-03:
Hello,

Sorry for the late answer.

Is the file executable by the user running octoprint ?

Bajotaz, 2021-01-01:
Got the same problem as kd7eir

CrystalJP, 2021-02-24:
I am also having this issue, I can run it from cmd as any user and I think my path is correct.

CrystalJP, 2021-02-24:
"C:\Users\John-Paul\Desktop\OctoPrint\arduino-cli_0.9.0_Windows_64bit\arduino-cli.exe"

does this look correct?

Renaud11232, 2021-02-24:
It does look correct.

Can you provide the logs from OctoPrint as well as the output from the follwing command so I can try to see what's going on :

```
C:\Users\John-Paul\Desktop\OctoPrint\arduino-cli_0.9.0_Windows_64bit\arduino-cli.exe version --format json
```

Thanks

closed 2021-05-04 by github-actions[bot]