Renaud11232/OctoPrint-Marlin-Flasher #63: web ui stuck on wizard

felixmxr, 2020-12-25:
similar to https://github.com/Renaud11232/OctoPrint-Marlin-Flasher/issues/32
screenshot
[image]

octoprint log
` octoprint.util.comm - INFO - Connecting to port /dev/ttyAMA0, baudrate 250000
2020-12-25 21:41:34,767 - octoprint.util.comm - INFO - Changing monitoring state from "Opening serial connection" to "Connecting"
2020-12-25 21:41:34,790 - octoprint.server.util.watchdog - INFO - Running initial scan on watched folder...
2020-12-25 21:41:34,794 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2020-12-25 21:41:34,803 - octoprint.server.util.watchdog - INFO - ... initial scan done.
2020-12-25 21:41:35,030 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2020-12-25 21:41:35,058 - octoprint.util.comm - INFO - Changing monitoring state from "Connecting" to "Operational"
2020-12-25 21:41:35,141 - octoprint.plugin - ERROR - Error while calling plugin marlin_flasher
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/__init__.py", line 271, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_marlin_flasher/__init__.py", line 100, in on_event
    self.__flasher.handle_connected_event()
AttributeError: 'MarlinFlasherPlugin' object has no attribute '_MarlinFlasherPlugin__flasher'
2020-12-25 21:41:35,248 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2020-12-25 21:41:35,309 - octoprint.util.comm - INFO - Printer reports firmware name "Marlin 2.0.5.3 (GitHub)"
2020-12-25 21:41:35,371 - octoprint.plugins.ABL_Expert - INFO - self.printer_cap[eeprom] is now 1
`

Renaud11232, 2020-12-26:
Hello, I have a few questions

In the web console, are there any errors before the ones you showed ?

Did you set the path to your arduinocli executable ?

Is there any error shown on the web page itself. Like a little red notification. And what does it say ?

Thanks

andymk3, 2020-12-29:
Same issue here. I had to remove the plugin via SSH to use my printer again with this:

`~/oprint/bin/pip uninstall OctoPrint-Marlin-Flasher`

Renaud11232, 2021-01-20:
I managed to reproduce the problem.

It looks like the issue comes from another plugin that prevents the Setup Wizard from saving settings.
@andymk3 Can you confirm you had the plugin `PSU Control` installed when you tried to install my plugin ?

I can see from the screenshot that @felixmxr had it installed.

felixmxr, 2021-01-20:
i have removed psu control and now your plugin works
thanks for helping.
sorry for not responding previously. i couldnt find any helpful info earlier

Renaud11232, 2021-02-01:
This issue has been fixed in `PSU Control 0.1.11`
[OctoPrint-SmartPreheat](https://github.com/kantlivelong/OctoPrint-SmartPreheat) is also affected and a fix will be released shortly.

I'm closing this since the cause issue is now fixed

closed 2021-02-01 by Renaud11232