Renaud11232/OctoPrint-Marlin-Flasher #64: Cannot exit the installation menu

creatron, 2020-12-30:


creatron, 2020-12-30:
I installed the plugin, and on reset you need to configure it, wanted to configure it later, as I have not yet installed Arduino not PlatformIO, I pressed yes to exit, but nothing happens.  now it is stucked and does not exit the menu, render Octoprint useless, couls also not uninstall it with pip, some incompatiblities. Octoprint yml  also changed so I could not enter safe mode.

Renaud11232, 2020-12-30:
Hello,

This seems to be the same issue as #63 

In the web console, are there any errors before the ones you showed ?

Is there any error in the console that appears when you click "Finish" ?

Did you set the path to your arduinocli executable ?

Is there any error shown on the web page itself. Like a little red notification. And what does it say ?

ShaneDelmore, 2021-01-19:
I have the same issue. Have only accessed my octopi from iOS at this point so I don't have access to web console but here is what I see:  
[image]

ShaneDelmore, 2021-01-19:
Accessing from chrome on my macbook I see: 
[image]

Renaud11232, 2021-01-19:
Hello,

Thanks for the screenshot.
Since I can't reproduce the issue on my side, I'll need some more information.

Do you have other plugins installed ? What plugins ?

Can you upload the whole file located at [http://octopi.local/static/webassets/packed_plugins.js](http://octopi.local/static/webassets/packed_plugins.js) here ?

Does the error `"Uncaught TypeError: Cannot read property 'version' of null` shows up when you click "Finish" ?
Is there any other error appearing when you click "Finish" ?

Thanks for you patience.

ShaneDelmore, 2021-01-19:
Packed_plugins.js here, renamed to txt file to allow uploading.

I am running latest release candidate of octopi on a raspberry pi 4.

I have a ton of plugins, my printer mainboard is damaged at the moment so I was just downloading and checking out many different plugins. This also means my printer is not connected in case that is useful info. Is there an easy way to generate a list?  Can I just ssh into the server and create the plugin list somehow?

[packed_plugins.txt](https://github.com/Renaud11232/OctoPrint-Marlin-Flasher/files/5837761/packed_plugins.txt)

The error appears before I hit finish.  When I hit finish I see this in the dev console: 
[image]

ShaneDelmore, 2021-01-19:
There is also an error loading sourcemap, so I don't know how to see what source code is causing the issue.  
[image]

ShaneDelmore, 2021-01-19:
I don't know a thing about js development, so not sure if this is useful but the developer tools shows the stack trace terminating at line 38 (o.apply(e, arguments)) of this formatted snippet of code:

`;!function() {
    function e(t, e) {
        return "function" != typeof t || t.replaced ? t : (e.replaced = !0,
        e)
    }
    var n, r, o, u = 0;
    if ("undefined" != typeof CustomEvent && "function" == typeof window.dispatchEvent) {
        function s(t) {
            try {
                if ("object" == typeof t && (t = JSON.stringify(t)),
                "string" == typeof t)
                    return window.dispatchEvent(new CustomEvent("lprequeststart",{
                        detail: {
                            data: t,
                            requestID: ++u
                        }
                    })),
                    u
            } catch (t) {}
        }
        function c(t) {
            try {
                window.dispatchEvent(new CustomEvent("lprequestend",{
                    detail: t
                }))
            } catch (t) {}
        }
        "undefined" != typeof XMLHttpRequest && XMLHttpRequest.prototype && XMLHttpRequest.prototype.send && (XMLHttpRequest.prototype.send = e(XMLHttpRequest.prototype.send, (o = XMLHttpRequest.prototype.send,
        function(t) {
            var e = this
              , n = s(t);
            return n && e.addEventListener("loadend", function() {
                c({
                    requestID: n,
                    statusCode: e.status
                })
            }),
            o.apply(e, arguments)
        }
        ))),
        "function" == typeof fetch && (fetch = e(fetch, (r = fetch,
        function(t, e) {
            var n = s(e)
              , o = r.apply(this, arguments);
            if (n) {
                function u(t) {
                    c({
                        requestID: n,
                        statusCode: t && t.status
                    })
                }
                o.then(u).catch(u)
            }
            return o
        }
        )));
        function i(t) {
            return e(t, function() {
                try {
                    this.dispatchEvent(new CustomEvent("lpsubmit"))
                } catch (t) {}
                return t.apply(this, arguments)
            })
        }
        function t() {
            if (document && document.forms && 0 < document.forms.length)
                for (var t = 0; t < document.forms.length; ++t)
                    document.forms[t].submit = i(document.forms[t].submit)
        }
        document && "interactive" === document.readyState || "complete" === document.readyState ? t() : window.addEventListener("DOMContentLoaded", t, !0),
        Document.prototype.createElement = e(Document.prototype.createElement, (n = Document.prototype.createElement,
        function() {
            var t = n.apply(this, arguments);
            return t && "FORM" === t.nodeName && t.submit && (t.submit = i(t.submit)),
            t
        }
        ))
    }
}();
`

ShaneDelmore, 2021-01-19:
The errors of not being able to parse 200C and 50C appear to be coming from this function:

```js
                    cb: function(b, c, d) {
                        switch (a.a.R(b)) {
                        case "option":
                            "string" === typeof c ? (a.a.g.set(b, a.c.options.$b, n),
                            "__ko__hasDomDataOptionValue__"in b && delete b.__ko__hasDomDataOptionValue__,
                            b.value = c) : (a.a.g.set(b, a.c.options.$b, c),
                            b.__ko__hasDomDataOptionValue__ = !0,
                            b.value = "number" === typeof c ? c : "");
                            break;
                        case "select":
                            if ("" === c || null === c)
                                c = n;
                            for (var e = -1, f = 0, g = b.options.length, h; f < g; ++f)
                                if (h = a.w.M(b.options[f]),
                                h == c || "" === h && c === n) {
                                    e = f;
                                    break
                                }
                            if (d || 0 <= e || c === n && 1 < b.size)
                                b.selectedIndex = e,
                                6 === a.a.W && a.a.setTimeout(function() {
                                    b.selectedIndex = e
                                }, 0);
                            break;
                        default:
                            if (null === c || c === n)
                                c = "";
                            b.value = c
                        }
                    }
```

Renaud11232, 2021-01-19:
> Packed_plugins.js here, renamed to txt file to allow uploading.
>
> I am running latest release candidate of octopi on a raspberry pi 4.
>
> I have a ton of plugins, my printer mainboard is damaged at the [...]

Thanks a lot, this will be very useful.
Actually, I can get the plugin list from the .js file

> There is also an error loading sourcemap, so I don't know how to see what source code is causing the issue.
> [image]

This is more of a warning, coming from a Chrome extension, nothing to worry about

> I don't know a thing about js development, so not sure if this is useful but the developer tools shows the stack trace terminating at line 38 (o.apply(e, arguments)) of this snippet of code:
> 
> ```js
> ;!function() {
>     function e(t, e) {
>         return "function" != typeof t || t.replaced ? t : (e.replaced = !0,
>         e)
>     }
>     var n, r, o, u = 0;
>     if ("undefined" != typeof CustomEvent && "function" == typeof window.dispatchEvent) {
>         function s(t) {
>             try {
>                 if ("object" == typeof t && (t = JSON.stringify(t)),
>                 "string" == typeof t)
>                     return window.dispatchEvent(new CustomEvent("lprequeststart",{
>                         detail: {
>                             data: t,
>                             requestID: ++u
>                         }
>                     })),
>                     u
>             } catch (t) {}
>         }
>         function c(t) {
>             try {
>                 window.dispatchEvent(new CustomEvent("lprequestend",{
>                     detail: t
>                 }))
>             } catch (t) {}
>         }
>         "undefined" != typeof XMLHttpRequest && XMLHttpRequest.prototype && XMLHttpRequest.prototype.send && (XMLHttpRequest.prototype.send = e(XMLHttpRequest.prototype.send, (o = XMLHttpRequest.prototype.send,
>         function(t) {
>             var e = this
>               , n = s(t);
>             return n && e.addEventListener("loadend", function() {
>                 c({
>                     requestID: n,
>                     statusCode: e.status
>                 })
>             }),
>             o.apply(e, arguments)
>         }
>         ))),
>         "function" == typeof fetch && (fetch = e(fetch, (r = fetch,
>         function(t, e) {
>             var n = s(e)
>               , o = r.apply(this, arguments);
>             if (n) {
>                 function u(t) {
>                     c({
>                         requestID: n,
>                         statusCode: t && t.status
>                     })
>                 }
>                 o.then(u).catch(u)
>             }
>             return o
>         }
>         )));
>         function i(t) {
>             return e(t, function() {
>                 try {
>                     this.dispatchEvent(new CustomEvent("lpsubmit"))
>                 } catch (t) {}
>                 return t.apply(this, arguments)
>             })
>         }
>         function t() {
>             if (document && document.forms && 0 < document.forms.length)
>                 for (var t = 0; t < document.forms.length; ++t)
>                     document.forms[t].submit = i(document.forms[t].submit)
>         }
>         document && "interactive" === document.readyState || "complete" === document.readyState ? t() : window.addEventListener("DOMContentLoaded", t, !0),
>         Document.prototype.createElement = e(Document.prototype.createElement, (n = Document.prototype.createElement,
>         function() {
>             var t = n.apply(this, arguments);
>             return t && "FORM" === t.nodeName && t.submit && (t.submit = i(t.submit)),
>             t
>         }
>         ))
>     }
> }();
> ```

> The errors of not being able to parse 200C and 50C appear to be coming from this function:
>
> ```js
>                     cb: function(b, c, d) {
>                         switch (a.a.R(b)) [...]

Thanks for these as well :)
I'll investigate and try to recreate the same environment as you.
I feel like this might be an issue with another plugin, maybe a conflict between mine and another plugin.

In the meantime, if you want to uninstall my plugin so it unlocks the OctoPrint interface, you can do it from the command line
```bash
~/oprint/bin/pip uninstall OctoPrint-Marlin-Flasher
```
I'll try to keep this thread up to date

Renaud11232, 2021-01-19:
I almost forgot,
Can you also send the logs

These are usualy under `~/.octoprint/logs`. The latest will be `octoprint.log`

Thanks

ShaneDelmore, 2021-01-19:
Here is my latest log. I uninstalled the plugin, so let me know if you need another log.  Also, if it would speed your work to pair with me on this I could setup some type of screen share and we could walk through it together.

[octoprint.log](https://github.com/Renaud11232/OctoPrint-Marlin-Flasher/files/5838088/octoprint.log)

ShaneDelmore, 2021-01-19:
Also, it's no rush on my end, I'm not in need of this at the moment, just noticed that this has been a thorn in your side for over a year and thought I would see if I could help with my duplication of it.

ShaneDelmore, 2021-01-19:
I think it may be a conflict with the "PSU Control" plugin from https://github.com/kantlivelong/OctoPrint-PSUControl/archive/master.zip

Try to see if you can reproduce by first installing PSU Control, then Marlin Flasher.

Renaud11232, 2021-01-20:
Hello,

Thanks for your help,
I managed to reproduce the issue.
And indeed, it looks like this is linked to the `PSU Control` plugin.

The same problem happens with other plugins showing a Wizard after installation, not just mine.
I tried to install PSU Control, then any plugin with a wizard setup (I tried Kronos Data Collector & Octprint-Anywhere). In both cases, I faced the same problem. I was not able to get over the setup screen.

I'll open an issue on PSU Control's repository.

@creatron Can you confirm that you had `PSU Control` installed when you tried to install Marlin-Flashed ?

ShaneDelmore, 2021-01-21:
If this happens with any other plugin that has a wizard then this spawns two new questions for me.

1.  Does it happen without Marlin-Flasher or PSUControl, just two random other plugins that have wizards?
2. If so, would that mean that this is actually an OctoPrint bug?

Renaud11232, 2021-01-21:
Hello,

From what I saw, it also happens without Marlin-Flasher, but only when PSUControl is installed.

Renaud11232, 2021-02-01:
This issue has been fixed in `PSU Control 0.1.11`
[OctoPrint-SmartPreheat](https://github.com/kantlivelong/OctoPrint-SmartPreheat) is also affected and a fix will be released shortly.

I'm closing this since the cause issue is now fixed

closed 2021-02-01 by Renaud11232