Renaud11232/OctoPrint-Marlin-Flasher #66: Feature request:  platformio installer

jrmoserbaltimore, 2021-01-31:
Simple enough:  build in a platformio installer/updater to create a virtual environment maintained by the plugin itself.

Renaud11232, 2021-01-31:
Hello,

While this would be possible, it is actually forbidden.
[On this page](https://plugins.octoprint.org/help/registering/) there is : 

> Your plugin doesn’t attempt to modify the user’s system without their knowledge, e.g. by trying to install additional system packages, services or the like. If your plugin needs additional steps like this to function, add a wizard dialog that prompts the user to do these things, do not do them automatically.

> Exception: Fetching additional Python dependencies from the Python Package Index through plugin_requires in your setup.py is fine.

Renaud11232, 2021-06-02:
I asked this question [on Discord](https://discord.com/channels/704958479194128507/708230829050036236/849638982262194218).
It is in fact allowed if it's made clear to the user so I'm reopening this

Renaud11232, 2022-01-15:
This is now available in the version 1.0.0

closed 2022-01-15 by Renaud11232