Renaud11232/OctoPrint-Marlin-Flasher #69: What exactly is the "Platform.io Project" that I am uploading?

Aylmayo, 2021-02-22:
I'm not the most familiar with all of this, I can edit Marlin through Atom and Platformio and that's about it 

I used to flash my board by dropping the firmware.bin onto the SD card and rebooting the printer, the Marlin Flasher calls for me to upload the Platform.io project "Make sure it's a zip file containing a valid Platform.io project for your board"

I'm unsure what I'm uploading to Marlin Flasher, do I make my own zip? And if so what does it contain?

Renaud11232, 2021-02-22:
Hello,

When using Platform.io, you should upload a zip file (that you will probably have to create yourself) containing the whole project.
What I mean by project is the whole folder that contains the `platformio.ini` file (and all the other files).

Please tell me if that helped or if you have more questions

Aylmayo, 2021-02-22:
Thank you so much for your quick response, so I just open Marlin 2.0 in Atom/Platformio, make my changes, build it, then go to the project folder and put all of this into a WinrRAR and then I can upload that to Marlin Flasher?

[image]

Renaud11232, 2021-02-22:
The plugin will compile the firmware so you don't need to build it first, this may take a while depending on the machine running OctoPrint

closed 2021-05-02 by github-actions[bot]