Renaud11232/OctoPrint-Marlin-Flasher #73: [BUG] Flashing failed

FrancYescO, 2021-04-14:
-Upgraded to latest version
-Upgraded arduino-cli to 0.17.0 (i was using 0.6.0)
-Uploaded a sketch that was working on the old versione
-"Flashing failed" message appear after few seconds after "Compiling"

how to get more debug infos?

[image]

Renaud11232, 2021-04-15:
Hello,

In OctoPrint settings, under the Logging section, you need to set the logging level to "DEBUG" for the plugin and then save :
[image]

Once this is done, you can redo the same operation, so it appears in the logs and link them in this issue.
Can you also send the zip file you uploaded ?

Thanks

FrancYescO, 2021-04-15:
nvm, it was just failing compiling due to a missing library, i just miss out the old popup where it show more detailed information about the failing lines ;)

Renaud11232, 2021-04-15:
I'm reopening since it should show the popup.

Can you send the zip file so I can see what happened ?

FrancYescO, 2021-04-15:
[Marlin.zip](https://github.com/Renaud11232/OctoPrint-Marlin-Flasher/files/6321061/Marlin.zip)
here you go

Renaud11232, 2021-04-15:
I think I found the issue, it seems to be an issue with `arduino-cli`.
Even when the compilation fails, arduino-cli does return an empty error output.
```json
{
  "compiler_out": "",
  "compiler_err": "",
  "builder_result": {
    "build_path": "C:\\Users\\gaspa\\AppData\\Local\\Temp\\arduino-sketch-1AF59F1FD8A66CA6FD1C3D0031EDAE86"
  },
  "success": false
}
```
I opened an issue a while ago for a similar problem,
I'll try to adapt my plugin until it's fixed in arduino-cli

FrancYescO, 2021-04-16:
For me to get some debug info I just stole with ps the command that marlin flasher was issuing, and after issuing the command on the terminal I noticed the compile error output (that was echoed before the JSON dictionary)

fritzw, 2021-05-26:
I'd suggest to add more helpful output to the popup in case the compilation or flashing fails. For example, I got the following popup which contains no information at all. Not even a hint, how to proceed  to get more relevant information.
[image]

When enabling the debug log as explained above, I got the following, which contains the actual problem (PlatformIO outdated). 

Maybe it could help to show at least the first 100 lines of stdout in the error message popup when an error occurs (limited to a fixed value in order to prevent excessive memory usage during compilation)?

```
[...]
2021-05-25 16:42:25,421 - octoprint.plugins.marlin_flasher - INFO - Compiling...
2021-05-25 16:42:25,423 - octoprint.plugins.marlin_flasher - DEBUG - Executing command : /home/pi/.platformio/penv/bin/platformio run -d /home/pi/.octoprint/data/marlin_flasher/firmware_platformio/Marlin
2021-05-25 16:42:28,864 - octoprint.plugins.marlin_flasher - DEBUG - Return code : 1
2021-05-25 16:42:28,865 - octoprint.plugins.marlin_flasher - DEBUG - Standard output : Processing melzi (platform: atmelavr; board: sanguino_atmega1284p; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option

**************************************************
******      An update to PlatformIO is      ******
******  required to build Marlin Firmware.  ******
******                                      ******
******      Minimum version:  (5, 0, 3)     ******
******      Current Version:  (4, 3, 4)     ******
******                                      ******
******   Update PlatformIO and try again.   ******
**************************************************
2021-05-25 16:42:28,865 - octoprint.plugins.marlin_flasher - DEBUG - Error output : ========================== [FAILED] Took 1.85 seconds ==========================
Environment               Status    Duration
------------------------  --------  ------------
mega2560                  IGNORED
mega1280                  IGNORED
MightyBoard1280           IGNORED
[...]
```

Renaud11232, 2021-05-27:
@fritzw I've opened a separate issue for that problem (#76) since this issue is related to an issue with arduino-cli

commit 1e896dea (2021-06-20): restores arduino flash functionality
fixes an issue where the error output would not show up because of an issue with arduino-cli
fixes #73
closed 2022-01-15 by Renaud11232