Renaud11232/OctoPrint-Marlin-Flasher #77: PlatformIO flashing failure.

AndKe, 2021-06-04:
**Describe the bug**
Error: Please specify `upload_port` for environment or use global `--upload-port` option.
For some development platforms it can be a USB flash drive (i.e. /media/<user>/<device name>)
*** [upload] Explicit exit, status 1
========================= [FAILED] Took 24.75 seconds =========================
Environment    Status    Duration
-------------  --------  ------------
LPC1768        FAILED    00:00:24.749
==================== 1 failed, 0 succeeded in 00:00:24.749 ====================

**To Reproduce**

Steps to reproduce the behavior:
1. upload a huge PlatformIO project
2. wait for a long time while it's compiled on RPi
3. observe fail

**Expected behavior**

It would be nice if one could upload a compiled .bin file to the plugin, which could just copy it as a file to the board, and reboot for self-flashing. 

**Version of OctoPrint-Marlin-Flasher**
Current release, from OctoPi plugin repository

Renaud11232, 2021-06-04:
Hello,

What board were you trying to flash ?
Could you provide the Marlin source you were trying to compile ?

For some boards (ie the SKR 1.3), you will need an additional piece of software as the boards appear as a usb drive. If this is your case : Did you install udevil according to [the wiki](https://github.com/Renaud11232/OctoPrint-Marlin-Flasher/wiki/PlatformIO-Installation#armmbed-boards) ?

Also, I've already gotten issues for adding a way to directly flash a compiled firmware without the need of compiling on the pi, but I don't think there is a way to do this with platformio. The closest way would be to add support for remote uploads (#71).

Thanks

AndKe, 2021-06-04:
@Renaud11232  I am pretty sure I missed that step, yes I use SKR1.3 (it's printing now after completing the update using SD-card.)
Thank you.

closed 2021-06-04 by AndKe