Renaud11232/OctoPrint-Marlin-Flasher #78: trying to upload hex file but getting "arduino cli version is not supported"

mark-hahn, 2021-07-01:
I want to do something very simple.  Just upload a compiled hex file from platformio on windows and then flash it.  When I try I get "arduino cli version is not supported".  Why do I need any cli?  Can't it just send the hex data over the usb connection?

Renaud11232, 2021-07-03:
Hello,

> When I try I get "arduino cli version is not supported"

The message "arduino-cli version is not supported" means that the version of arduino-cli you have installed is not supported by the plugin.
Currently, Marlin-Flasher only supports versions from 0.15 up to 0.17. This means that the path to arduino-cli you have set in the settings does not point to arduino-cli between 0.15 and 0.17.
With version 0.5.0 of the plugin, I dropped support for older versions of arduino-cli, this might be why this message shows up. So you need to update arduino-cli first.

> Why do I need any cli? Can't it just send the hex data over the usb connection?

Not all board use the same method for the flashing process, and often require specific tools. Arduino-cli (as well as platformio) handle the upload to the board using the right tool for each board. That way, I don't need to write specific code for each board and makes the plugin compatible with most boards.

If the board you are trying to flash is compatble with arduino-cli, then you should be able to directly flash a compiled .hex file.
However, if your board is only compatible with PlatformIO and not with arduino-cli, then you can't directly use the .hex file.

Let me know if it helped.

mark-hahn, 2021-07-03:
Thanks.  I will look into this.  I have been loading directly from platformio on windows but dragging my computer to the garage has not been fun.

closed 2021-09-09 by github-actions[bot]