Renaud11232/OctoPrint-Marlin-Flasher #81: Octoprint Marlin Flasher Platformio

JBitsik, 2021-12-08:
Hello ,
I 've been trying hard to find how to flash Marlin to my print as a PlatformIO project.
I searched for videos or written detailed instructions but nothing usefull came up.
I have read your instructions at 
https://github-wiki-see.page/m/Renaud11232/OctoPrint-Marlin-Flasher/wiki_index
but I think there are some missing details for people like who do not know many stuff about these things.
I installed PlatformIO in Linux as described.
Set the path to Platformio-Core at /opt/pio/bin/pio
What a zip file containing a valid PlatformIO means?
Which files should I contain? For example I made a zip containing the platformio.ini-Configuration.h-Configuration_adv.h - Is it right?
I tried to flash but there it says "warning: Calling missing SConscript without error is deprecated.
Transition by adding must_exist=False to SConscript calls.
Missing SConscript 'buildroot/share/PlatformIO/scripts/common-dependencies.py'
File "/opt/pio/lib/python3.7/site-packages/platformio/builder/main.py", line 178, in <module>

warning: Ignoring missing SConscript 'buildroot/share/PlatformIO/scripts/common-cxxflags.py'
File "/opt/pio/lib/python3.7/site-packages/platformio/builder/main.py", line 178, in <module>
Error: Nothing to build. Please put your source code files to '/home/pi/.octoprint/data/marlin_flasher/firmware_platformio/Marlin' folder
========================== [FAILED] Took 2.08 seconds ==========================
Environment    Status    Duration
-------------  --------  ------------
LPC1769        FAILED    00:00:02.077
==================== 1 failed, 0 succeeded in 00:00:02.077 ===================="
Sorry for my big message , I would appreciate if you could help me.

Kind regards 
John

Renaud11232, 2021-12-11:
Hello,

The fact that the plugin did not give any errors before you tried to actually flash the firmware means you probably set it up right. No issue there...

The zip file should not only contain the configuration files, but the whole Marlin project. (Including all source files)

I hope this made it more clear

Please tell me if that worked

JBitsik, 2021-12-11:
Hi , thanks for your response.
One thing only because I am a newbie.
What do you mean the whole Marlin project? Which folders it must contain?
Can you help me , I searched a lot in the internet and there is no clear solution.

[image]

Mordi1984, 2021-12-11:
+1 I need an manual too. I tried almost everything in VS Code. Where exactly can i generate the ZIP file? Manual or Automatic?

Mordi1984, 2021-12-11:
Okay i get it.
I zipped the whole folder with the edited (In Virtual Studio Code) files.
[image]

when i opened it, it is the whole folder:
[image]

the folder is the whole firmware with platformio things
(the files where added by plattformio, when you inserted the Firmware from Marlin in Visual Studio Code on the Marlin Tab (like if you want to build it.)

If you load the whole Zip File with your configuration for your Printer or made by yourself into octoprint, he will let you choose which USB Device you choose, and your favour of building. 
The build will produce on the device where you ran octoprint.

I thought before, that i need to compile/build it on my Computer first.
But no that will do the raspi then.

Beare after upload you need to reset the eeprom settings.
On Octoprint Terminal it is M500.
Then restart your 3D Printer.

If you have any Probe, you need to do calibration first, and safe it after then.

Then you are ready to go.

Hopefully i dont have made any foults. But in my case it worked perfekt.

But yes, in the Manual that would be very helpfully if it would be clear for idiots like me, who think complicated.

After that enjoy the Tool, it is very good then.

Mordi1984, 2021-12-11:
[image]

Renaud11232, 2021-12-13:
Hello,

Sorry for the late answers

I'll try to add a section in the wiki on how to create that zip file.

Indeed you need to create a zip file with all of the contents of your Marlin folder (the one that contains `platformio.ini`) and all of its subfolders.
To create the zip file you can right-click on the folder to zip > Send to > Compressed folder (not sure about the exact translation) or you can use software like 7zip (on Windows)

The compilation will indeed take place on the machine that hosts Octoprint (for most users that will be a Raspberry Pi). This is a slow process but I don't know any other way using PlatformIO.

I hope that helps

Renaud11232, 2021-12-19:
A section on how to create the zip file has been added to the wiki, I'm closing this

closed 2021-12-19 by Renaud11232