Renaud11232/OctoPrint-Marlin-Flasher #84: [BUG]

broncosis, 2022-03-25:
**Describe the bug**

the Arduino install fails on RPI-3 

Unable to detect architecture : armv6l

**To Reproduce**

Steps to reproduce the behavior:
1. load clean image of octopi
2. install this plugin 
3. and reboot 
4. then atempt to install arduino 

**Expected behavior**

it should install the ardunio-cli

**Screenshots**

If applicable, add screenshots to help explain your problem.

Marlin flasher 1.01

[octoprint.log](https://github.com/Renaud11232/OctoPrint-Marlin-Flasher/files/8347432/octoprint.log)

using Firefox

broncosis, 2022-03-25:
[octoprint(1).log](https://github.com/Renaud11232/OctoPrint-Marlin-Flasher/files/8347442/octoprint.1.log)
 updated log with debugging turned on

Renaud11232, 2022-03-25:
Hello,

Thanks for reporting this.
Starting from v1.0.2 (just published now), you should be able to install arduino-cli for `armv6l` as well.

closed 2022-03-25 by Renaud11232

broncosis, 2022-03-26:
works great thanks

commit 660962c0 (2022-03-25): fixes #84
also fixes an error showing in the logs when arduino-cli is not setup yet