Renaud11232/OctoPrint-Marlin-Flasher #86: Unable to find destination disk (Autodetect Error)

Naesstrom, 2022-07-28:
I'm trying to flash an BTT SKR E3 Turbo and I installed the udevil and started it, but now when I try to flash it can't find the destination disk. Do I need to change something in the platformio file to?

`Warning! Ignore unknown configuration option `monitor_flags` in section [env]
Processing LPC1769 (board: nxp_lpc1769; platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.3.zip; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1769.html
PLATFORM: NXP Arduino LPC176x (0.1.3) > NXP LPC1769
HARDWARE: LPC1769 120MHz, 31.97KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES: 
 - framework-arduino-lpc176x @ 0.2.9 
 - toolchain-gccarmnoneeabi @ 1.100301.220327 (10.3.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ strict
Found 3 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Servo @ 1.0.0
|-- TMCStepper @ 0.7.3
|-- U8glib-HAL @ 0.5.2
Building in release mode
Compiling .pio/build/LPC1769/src/src/inc/Warnings.cpp.o
Marlin/src/inc/Warnings.cpp:195:6: warning: #warning "Note: Auto-assigned Z2 STEP/DIR/ENABLE_PINs to unused En_STEP/DIR/ENABLE_PINs. (Define NO_AUTO_ASSIGN_WARNING to suppress this warning.)" [-Wcpp]
  195 |     #warning "Note: Auto-assigned Z2 STEP/DIR/ENABLE_PINs to unused En_STEP/DIR/ENABLE_PINs. (Define NO_AUTO_ASSIGN_WARNING to suppress this warning.)"
      |      ^~~~~~~
Marlin/src/inc/Warnings.cpp:728:6: warning: #warning "Motherboard DIAG jumpers must be removed when SENSORLESS_HOMING is disabled. (Define DIAG_JUMPERS_REMOVED to suppress this warning.)" [-Wcpp]
  728 |     #warning "Motherboard DIAG jumpers must be removed when SENSORLESS_HOMING is disabled. (Define DIAG_JUMPERS_REMOVED to suppress this warning.)"
      |      ^~~~~~~
Checking size .pio/build/LPC1769/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [===       ]  32.9% (used 10780 bytes from 32736 bytes)
Flash: [=====     ]  47.6% (used 225928 bytes from 475136 bytes)
before_upload(["upload"], [".pio/build/LPC1769/firmware.bin"])

Unable to find destination disk (Autodetect Error)
Please select it in platformio.ini using the upload_port keyword (https://docs.platformio.org/en/latest/projectconf/section_env_upload.html) or copy the firmware (.pio/build/LPC1769/firmware.bin) manually to the appropriate disk

Configuring upload protocol...
AVAILABLE: blackmagic, cmsis-dap, jlink, mbed
CURRENT: upload_protocol = mbed
Looking for upload disk...
Error: Please specify `upload_port` for environment or use global `--upload-port` option.
For some development platforms it can be a USB flash drive (i.e. /media/<user>/<device name>)
*** [upload] Explicit exit, status 1
========================= [FAILED] Took 15.17 seconds =========================

Environment    Status    Duration
----
---------  --------  ------------
LPC1769        FAILED    00:00:15.167
==================== 1 failed, 0 succeeded in 00:00:15.167 ====================`

Here's the file I'm trying to flash: [Marlin-2.1.zip](https://github.com/Renaud11232/OctoPrint-Marlin-Flasher/files/9206969/Marlin-2.1.zip)

Renaud11232, 2022-07-28:
Hello,

Can you provide the output of the following commands (assuming you are running on a RPi running octopi) :

```bash
ls -lah /mnt
ls -lah /media
systemctl status devmon@pi
```

If you are not using a RPi replace `devmon@pi` with `devmon@<the user running OctoPrint>` without the `<>`

Thanks

Naesstrom, 2022-07-28:
Yup here it is. I manually created the "pi" folder in the "Media" one earlier today when it said it was missing.
```
pi@octopi:~ $ ls -lah /mnt
total 8.0K
drwxr-xr-x  2 root root 4.0K Dec  2  2020 .
drwxr-xr-x 18 root root 4.0K Dec  2  2020 ..
pi@octopi:~ $ ls -lah /media
total 12K
drwxr-xr-x  3 root root 4.0K Jul 28 09:59 .
drwxr-xr-x 18 root root 4.0K Dec  2  2020 ..
drwxr-xr-x  2 root root 4.0K Jul 28 09:59 pi
pi@octopi:~ $ systemctl status devmon@pi
● devmon@pi.service - devmon Service
   Loaded: loaded (/lib/systemd/system/devmon@.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2022-07-28 10:03:34 CEST; 10h ago
 Main PID: 1943 (devmon)
    Tasks: 2 (limit: 4915)
   CGroup: /system.slice/system-devmon.slice/devmon@pi.service
           ├─1943 /bin/bash /usr/bin/devmon
           └─1964 /usr/bin/udevil --monitor

Jul 28 10:03:34 octopi systemd[1]: Started devmon Service.
Jul 28 12:55:33 octopi devmon[1943]: ===========================================
Jul 28 12:55:33 octopi devmon[1943]: Monitoring activity from the disks daemon. Press Ctrl+C to cancel.
Jul 28 12:55:33 octopi devmon[1943]: ===========================================
Jul 28 12:55:33 octopi devmon[1943]: removed:   /org/freedesktop/UDisks/devices/sda
Jul 28 12:55:33 octopi devmon[1943]: devmon: /usr/bin/udevil clean
pi@octopi:~ $
```
Might be related but I've tried uploading it through visual studio code to without success, says there's no port available even though I can connect with pronterface for example

Renaud11232, 2022-07-29:
Hello,

If I remember correctly, you don't need to create the `pi` directory manually as udevil will create it automatically once it detects a mass storage device.

I have other questions :

Is the SD Card usable by the version of Marlin currently running on the board (ie for storing gcode) ?
If so, you'll need to unmount it. (`M22` gcode command or via OctoPrint "eject" button)
If this fixes the issue, you can add this to the plugin settings so it does it automatically every time. (Settings > Marlin Falsher > General > Advanced options > Pre flash script > Type in M22). You may also need to add a small delay to let udevil mount the sd card.

When you tried directly with Vistual Studio Code, was it on Windows ?

Thanks

closed 2022-10-05 by github-actions[bot]