Renaud11232/OctoPrint-Marlin-Flasher #87: [BUG] Default file upload limit renders plugin unusable

orrinwitt, 2022-08-15:
Unless I'm misunderstanding something...

I'm very confused, because the full project when zipped with the current Marlin firmware (2.1.1) is around 83mb. With an upload size limit of 20mb, that makes it impossible to upload the project. 

3 questions:
1. Are there files I should be excluding from the zip?
2. Do I need to edit the configuration file from 20mb to something big enough?
3. Is there a reason for the 20mb limit that will cause problems if I more than quadruple the limit size?

orrinwitt, 2022-08-15:
ok, so based on the image in this closed issue,
it would seem i need to exclude the folders: .pio and .vscode
this renders a .zip file under 10mb

is that correct?

Renaud11232, 2022-08-15:
Hello,

Yes, the zip file should not contain these files.
`.vscode` is specific to Visual Studio code and won't be used
`.pio` contains various files including the compiled firmware (if you compiled it already) which will be ignored since Platform IO will recompile the project anyway.

orrinwitt, 2022-08-15:
Thank you for the help! I ended up running into another issue, which I reference here:
[](https://github.com/MarlinFirmware/AutoBuildMarlin/issues/61)

It seems to me that it's a platformio issue, but it affected your plugin as well. not sure if it's specific to my board or what, but I was surprised to see it affected me across 3 computers and 2 operating systems.

closed 2022-10-24 by github-actions[bot]