Renaud11232/OctoPrint-Marlin-Flasher #88: [BUG] Package failure

Getn67, 2022-09-12:
**Describe the bug**

Could not locate the correct package for plugin to function properly.

Rpi 3+ B
Octopi

**To Reproduce**

Steps to reproduce the behavior:

**Expected behavior**

necessary package installed to successfully flash printer

**Screenshots**

If applicable, add screenshots to help explain your problem.

**Version of OctoPrint-Marlin-Flasher**

1.0.3 installed via plugin manager

**Server configuration**
 - OS: https://github.com/guysoft/OctoPi (raspbian)
 - Browser Microsoft Edge
Version 105.0.1343.27 (Official build) (64-bit)
 - Python version Python 3.7.3

Tool Manager: Installing platformio/tool-dfuutil @ ^1.11.0
Error: Could not find the package with 'platformio/tool-dfuutil @ ^1.11.0' requirements for your system 'linux_armv7l'

Upload your logs here

**Browser console logs**

Upload the logs of your browser here, on Chrome press F12

Getn67, 2022-09-17:
1. when attempting to flash firmware
2. 
[image]
Which log would you like?
3. .18 the latest release.

Renaud11232, 2022-09-17:
You need to set `octoprint.plugins.marlin_flasher` to `DEBUG`. Once that is done, reproduce the error and download the `octiprint.log` file in `Settings > Logging` and upload it link it to this issue.
You can then unset the log level of `octoprint.plugins.marlin_flasher`

Getn67, 2022-09-18:
[octoprint (1).log](https://github.com/Renaud11232/OctoPrint-Marlin-Flasher/files/9593004/octoprint.1.log)

Hopefully this has everything you need.

Renaud11232, 2022-09-18:
Hello,

As I suspected, this is an issue with PlatformIO tools not being available for armv7l at a specific version, sadly, there's nothing I can do to fix that. It can however get fixed in the future, but I have no control over that.

Since this is not an issue with the plugin itself, I'll be closing this issue.

Sorry for the inconvenience

closed 2022-09-18 by Renaud11232

d4icon, 2023-01-31:
same problem here.
Raspberry pi 3, last version octoprint.

Any solution?