Renaud11232/OctoPrint-Marlin-Flasher #89: [BUG] Default Setting value for a list causes Octoprint log spam

synman, 2023-01-21:
**Describe the bug**

Periodically the octoprint log spits out the following:

```
2023-01-21 16:30:06,158 - octoprint.settings - WARNING - There is a non-dict value on the path to pluginsmarlin_flasherplatformiolast_flash_optionsenv at ['plugins', 'marlin_flasher', 'platformio', 'last_flash_options'], ignoring.
2023-01-21 16:30:07,018 - octoprint.settings - WARNING - There is a non-dict value on the path to pluginsmarlin_flasherplatformiolast_flash_optionsenv at ['platformio', 'last_flash_options'], ignoring.
2023-01-21 16:30:07,437 - octoprint.settings - WARNING - There is a non-dict value on the path to pluginsmarlin_flasherplatformiolast_flash_optionsenv at ['plugins', 'marlin_flasher', 'platformio', 'last_flash_options'], ignoring.
2023-01-21 16:30:08,256 - octoprint.settings - WARNING - There is a non-dict value on the path to pluginsmarlin_flasherplatformiolast_flash_optionsenv at ['platformio', 'last_flash_options'], ignoring.
2023-01-21 16:30:09,503 - octoprint.settings - WARNING - There is a non-dict value on the path to pluginsmarlin_flasherplatformiolast_flash_optionsenv at ['platformio', 'last_flash_options'], ignoring.
2023-01-21 16:30:09,896 - octoprint.settings - WARNING - There is a non-dict value on the path to pluginsmarlin_flasherplatformiolast_flash_optionsenv at ['plugins', 'marlin_flasher', 'platformio', 'last_flash_options'], ignoring.
2023-01-21 16:30:10,703 - octoprint.settings - WARNING - There is a non-dict value on the path to pluginsmarlin_flasherplatformiolast_flash_optionsenv at ['platformio', 'last_flash_options'], ignoring.
2023-01-21 16:30:11,824 - octoprint.settings - WARNING - There is a non-dict value on the path to pluginsmarlin_flasherplatformiolast_flash_optionsenv at ['platformio', 'last_flash_options'], ignoring.
2023-01-21 16:30:12,207 - octoprint.settings - WARNING - There is a non-dict value on the path to pluginsmarlin_flasherplatformiolast_flash_optionsenv at ['plugins', 'marlin_flasher', 'platformio', 'last_flash_options'], ignoring.
```

**To Reproduce**
```tail -F .octoprint/logs/octoprint.log
```

**Expected behavior**

clean log

**Screenshots**

N/A

**Version of OctoPrint-Marlin-Flasher**

Latest version installed yesterday via Plugin Manager

** final words **

thank you for such an awesome plugin!   

I've seen this error myself when you have a dict / list in your plugin settings that has a default value of None.  I believe a future version of Octoprint is slated to handle this condition a bit more gracefully but for now we get some log spam as a result of it.

Looks like this is the source of it (https://github.com/Renaud11232/OctoPrint-Marlin-Flasher/blob/master/octoprint_marlin_flasher/__init__.py#L39):

Changing to this should solve it:
```
			platformio=dict(
				cli_path=None,
				last_flash_options={}
			)
```

Renaud11232, 2023-01-22:
Hello and thanks for reporting this.

I noticed this a while ago but could find where it came from.

This is now fixed in 1.0.5

closed 2023-01-22 by Renaud11232

synman, 2023-01-23:
I chased my tail for a solid hour a couple weeks back getting to the heart of it when it bit me.

commit 84d2f0b0 (2023-01-22): Changes lest flash options default to emty dict
fixes #89