Renaud11232/OctoPrint-Marlin-Flasher #90: [FR] Support custom boards, or at least cpu/processor values

jaxzin, 2023-03-28:
**Is your feature request related to a problem? Please describe.**
My Melzi board needs to use the following fqbn value: Sanguino:avr:sanguino:cpu=atmega1284p
But I don't appear to have a way to append the cpu.

**Describe the solution you'd like**
The ability to enter the fqbn myself, or a GUI that supports appending the cpu value.

Renaud11232, 2023-03-28:
Hello,

This should be working, normally all the options should be displayed. If they are not, there might be a bug somewhere.

Could you please provide the logs (`octoprint.log`) so see if there is any error there ?

I'm guessing the plugin is configured to use arduino-cli in your case. Can you tell what version of arduino-cli is installed ?

Thanks

jaxzin, 2023-03-28:
I think this is a bit of user error. I first installed the Sanguino Core and was trying to Flash my device. That failed because I also needed the arduino:avr Core too. I didn't reload the interface at the time and while Sanguino appeared as an option under Board, the Processor field didn't appear. When I came back later the Processor field appeared under the Flash section and I was able to Flash my device successfully.

Renaud11232, 2023-03-29:
I'll close this issue since this is not a bug.

I'll look into automatically installing the `arduino:avr` core when installing arduino-cli through the settings.

closed 2023-03-29 by Renaud11232